Igalia/meta-webkit #511: scarthgap branch

fabioestevam, 2024-09-21:
Are there any plans to provide a scarthgap branch for meta-webkit?

Thanks

psaavedra, 2024-10-24:
Hi sorry for this belated response. Yes we have and we already did  as part of the tasks involved in https://github.com/Igalia/meta-webkit/pull/516.  The https://github.com/Igalia/meta-webkit/tree/scarthgap is now available. However, you notice that the main branch (Styhead) will be also compatible with the older releases (from kirkstone , included) as far as it will be possible. This is useful if you are interesting in getting latest stable versions of WPEWebKit  and other components (https://github.com/Igalia/meta-webkit/blob/main/conf/layer.conf#L20 ) for your build environment.

closed 2024-10-24 by psaavedra

psaavedra, 2024-10-24:
BTW, to reflect the main branch as compatible for the latest Yocto release, I've created [this Action](https://github.com/Igalia/meta-webkit/commit/c7955369ef604db6cf3d475941c18d3780f268fe). Therefore you get now main==styhead release branch but as I mentioned in my previous comment this branch keeps backward compatibility with the previous yocto releases up to kirkstone (LTS).