Igalia/meta-webkit #517: build fails - No package 'wpe-webkit-1.0' found

chrisvollorenew, 2024-10-29:
**Bug description**
Build with kirkstone fails - No package 'wpe-webkit-1.0' found

**How To Reproduce**
1. checkout top of kirstone branch
2. set these variables:
    PREFERRED_VERSION_cog = "0.12.4"
    PREFERRED_VERSION_wpewebkit = "2.36.8"
3. build

**Expected behavior**
build succeeds

**Screenshots**
`
ERROR: cog-0.12.4-r0 do_configure: ExecutionError('/home/chris/yocto-kirkstone-pocket2/build/tmp/work/cortexa7t2hf-neon-poky-linux-gnueabi/cog/0.12.4-r0/temp/run.do_configure.3853263', 1, None, None)
ERROR: Logfile of failure stored in: /home/chris/yocto-kirkstone-pocket2/build/tmp/work/cortexa7t2hf-neon-poky-linux-gnueabi/cog/0.12.4-r0/temp/log.do_configure.3853263
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/home/chris/yocto-kirkstone-pocket2/build/../meta-webkit/recipes-browser/wpebackend-fdo/wpebackend-fdo_1.14.1.bb:do_populate_sysroot', '/home/chris/yocto-kirkstone-pocket2/build/../meta-webkit/recipes-browser/wpewebkit/wpewebkit_2.36.8.bb:do_populate_sysroot', '/home/chris/yocto-kirkstone-pocket2/build/../poky/meta/recipes-core/glib-2.0/glib-2.0_2.72.3.bb:do_populate_sysroot', '/home/chris/yocto-kirkstone-pocket2/build/../poky/meta/recipes-core/glibc/glibc_2.35.bb:do_populate_sysroot', '/home/chris/yocto-kirkstone-pocket2/build/../poky/meta/recipes-devtools/cmake/cmake-native_3.22.3.bb:do_populate_sysroot', '/home/chris/yocto-kirkstone-pocket2/build/../poky/meta/recipes-devtools/gcc/gcc-cross_11.5.bb:do_populate_sysroot', '/home/chris/yocto-kirkstone-pocket2/build/../poky/meta/recipes-devtools/gcc/gcc-runtime_11.5.bb:do_populate_sysroot', '/home/chris/yocto-kirkstone-pocket2/build/../poky/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', '/home/chris/yocto-kirkstone-pocket2/build/../poky/meta/recipes-graphics/drm/libdrm_2.4.110.bb:do_populate_sysroot', '/home/chris/yocto-kirkstone-pocket2/build/../poky/meta/recipes-graphics/mesa/mesa_22.0.3.bb:do_populate_sysroot', '/home/chris/yocto-kirkstone-pocket2/build/../poky/meta/recipes-graphics/wayland/libinput_1.19.4.bb:do_populate_sysroot', '/home/chris/yocto-kirkstone-pocket2/build/../poky/meta/recipes-graphics/wayland/wayland-protocols_1.25.bb:do_populate_sysroot', '/home/chris/yocto-kirkstone-pocket2/build/../poky/meta/recipes-support/libsoup/libsoup-2.4_2.74.2.bb:do_populate_sysroot', '/home/chris/yocto-kirkstone-pocket2/build/../poky/meta/recipes-support/libsoup/libsoup_3.0.7.bb:do_populate_sysroot', 'virtual:native:/home/chris/yocto-kirkstone-pocket2/build/../poky/meta/recipes-devtools/ninja/ninja_1.10.2.bb:do_populate_sysroot', 'virtual:native:/home/chris/yocto-kirkstone-pocket2/build/../poky/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/home/chris/yocto-kirkstone-pocket2/build/../poky/meta/recipes-devtools/pkgconfig/pkgconfig_git.bb:do_populate_sysroot', 'virtual:native:/home/chris/yocto-kirkstone-pocket2/build/../poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/home/chris/yocto-kirkstone-pocket2/build/../poky/meta/recipes-extended/xz/xz_5.2.6.bb:do_populate_sysroot', 'virtual:native:/home/chris/yocto-kirkstone-pocket2/build/../poky/meta/recipes-graphics/wayland/wayland_1.20.0.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['wpebackend-fdo', 'wpewebkit', 'glib-2.0', 'glibc', 'cmake-native', 'gcc-cross-arm', 'gcc-runtime', 'quilt-native', 'libdrm', 'mesa', 'libinput', 'wayland-protocols', 'libsoup-2.4', 'libsoup', 'ninja-native', 'patch-native', 'pkgconfig-native', 'pseudo-native', 'xz-native', 'wayland-native', 'libtool-native', 'gettext-minimal-native', 'libpsl', 'gobject-introspection', 'sqlite3', 'libxml2', 'attr-native', 'linux-libc-headers', 'libcheck', 'mtdev', 'systemd', 'libevdev', 'libpcre', 'bash-completion', 'util-linux', 'libffi', 'zlib', 'libwpe', 'libepoxy', 'wayland', 'libxkbcommon', 'libgcc', 'icu', 'gstreamer1.0-plugins-base', 'woff2', 'libjpeg-turbo', 'freetype', 'harfbuzz', 'openjpeg', 'gstreamer1.0', 'gstreamer1.0-plugins-good', 'libgcrypt', 'libtasn1', 'libwebp', 'libxslt', 'cairo', 'fontconfig', 'libpng', 'expat-native', 'python3-native', 'libffi-native', 'libxml2-native', 'libxshmfence', 'libxext', 'expat', 'libxdamage', 'xrandr', 'libxfixes', 'xorgproto', 'libxxf86vm', 'libx11', 'libpthread-stubs', 'libpciaccess', 'mpfr-native', 'flex-native', 'gnu-config-native', 'binutils-cross-arm', 'gmp-native', 'zlib-native', 'libmpc-native', 'texinfo-dummy-native', 'zstd-native', 're2c-native', 'nghttp2', 'ncurses-native', 'curl-native', 'bzip2-native', 'python3', 'opkg-utils', 'autoconf-archive', 'base-files', 'libcap', 'shadow-native', 'base-passwd', 'libseccomp', 'kmod', 'acl', 'shadow', 'shadow-sysroot', 'zstd', 'bzip2', 'util-linux-libuuid', 'libxcrypt', 'libcap-ng', 'ncurses', 'xkeyboard-config', 'libxcb', 'libogg', 'libxv', 'libtheora', 'alsa-lib', 'libglu', 'pango', 'libsm', 'libopus', 'libvorbis', 'orc', 'iso-codes', 'libxrender', 'brotli', 'lcms', 'tiff', 'taglib', 'flac', 'libgudev', 'pulseaudio', 'gdk-pixbuf', 'mpg123', 'v4l-utils', 'libvpx', 'speex', 'wavpack', 'lame', 'libgpg-error', 'pixman', 'sqlite3-native', 'readline-native', 'util-linux-libuuid-native', 'gdbm-native', 'libtirpc-native', 'openssl-native', 'libnsl2-native', 'util-macros', 'libxrandr', 'xtrans', 'm4-native', 'libtirpc', 'openssl', 'xz', 'libnsl2', 'gdbm', 'readline', 'attr', 'xcb-proto', 'libxau', 'libxdmcp', 'libxft', 'fribidi', 'libice', 'ne10', 'libxtst', 'bluez5', 'ofono', 'libatomic-ops', 'libsndfile1', 'libtool', 'avahi', 'sbc', 'speexdsp', 'dbus', 'shared-mime-info', 'perl-native', 'libxi', 'libical', 'ell', 'mobile-broadband-provider-info', 'libdaemon', 'make-native', 'vala']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_configure
| -- The C compiler identification is GNU 11.5.0
| -- Detecting C compiler ABI info
| -- Detecting C compiler ABI info - done
| -- Check for working C compiler: /home/chris/yocto-kirkstone-pocket2/build/tmp/work/cortexa7t2hf-neon-poky-linux-gnueabi/cog/0.12.4-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-gcc - skipped
| -- Detecting C compile features
| -- Detecting C compile features - done
| -- Cog module path: /usr/lib/cog/modules
| -- Found PkgConfig: /home/chris/yocto-kirkstone-pocket2/build/tmp/work/cortexa7t2hf-neon-poky-linux-gnueabi/cog/0.12.4-r0/recipe-sysroot-native/usr/bin/pkg-config (found version "0.29.2")
| -- Checking for module 'gio-2.0>=2.44'
| --   Found gio-2.0, version 2.72.3
| -- Checking for module 'libsoup-2.4'
| --   Found libsoup-2.4, version 2.74.2
| -- Checking for module 'wpe-webkit-1.0>=2.28.0'
| --   No package 'wpe-webkit-1.0' found
| CMake Error at /home/chris/yocto-kirkstone-pocket2/build/tmp/work/cortexa7t2hf-neon-poky-linux-gnueabi/cog/0.12.4-r0/recipe-sysroot-native/usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:603 (message):
|   A required package was not found
| Call Stack (most recent call first):
|   /home/chris/yocto-kirkstone-pocket2/build/tmp/work/cortexa7t2hf-neon-poky-linux-gnueabi/cog/0.12.4-r0/recipe-sysroot-native/usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:825 (_pkg_check_modules_internal)
|   CMakeLists.txt:156 (pkg_check_modules)
| 
| 
| -- Configuring incomplete, errors occurred!
| See also "/home/chris/yocto-kirkstone-pocket2/build/tmp/work/cortexa7t2hf-neon-poky-linux-gnueabi/cog/0.12.4-r0/build/CMakeFiles/CMakeOutput.log".
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/chris/yocto-kirkstone-pocket2/build/../meta-webkit/recipes-browser/cog/cog_0.12.4.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 6772 tasks of which 6758 didn't need to be rerun and 1 failed.

`

**Environment (please complete the following information):**
 - yocto release: kirkstone

psaavedra, 2024-11-10:
The kirkstone branch is a bit out to date. However I would like to suggest you to use the `scarthgap` (latest LTS) branch what is also compatible with `kirkstone`(https://github.com/Igalia/meta-webkit/blob/scarthgap/conf/layer.conf#L20).

psaavedra, 2024-11-10:
Regarding the specific compilation error you sent, the failure is possibly due to the fact that the cog version 0.12.x does not support Soup2, so WebKit must be compiled with the option `-DUSE_SOUP2=ON` (https://github.com/Igalia/meta-webkit/blob/kirkstone/recipes-browser/wpewebkit/wpewebkit.inc#L87). This can be done by adding the `soup` option to the wpewebkit package.

### Example of adding "soup" flag to the wpewebkit package in `local.conf`:

```bash
PACKAGECONFIG:append:pn-wpewebkit = " soup"
```

chrisvollorenew, 2024-11-11:
@psaavedra thank you for this info! I tried adding this: 
`PACKAGECONFIG:append:pn-wpewebkit = " soup"`

And the build still fails like this
```
ERROR: cog-0.12.4-r0 do_configure: ExecutionError('/home/chris/yocto-meta-renew-kirkstone/build/tmp/work/cortexa7t2hf-neon-poky-linux-gnueabi/cog/0.12.4-r0/temp/run.do_configure.375992', 1, None, None)
ERROR: Logfile of failure stored in: /home/chris/yocto-meta-renew-kirkstone/build/tmp/work/cortexa7t2hf-neon-poky-linux-gnueabi/cog/0.12.4-r0/temp/log.do_configure.375992
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/home/chris/yocto-meta-renew-kirkstone/build/../meta-webkit/recipes-browser/wpebackend-fdo/wpebackend-fdo_1.14.1.bb:do_populate_sysroot', '/home/chris/yocto-meta-renew-kirkstone/build/../meta-webkit/recipes-browser/wpewebkit/wpewebkit_2.36.8.bb:do_populate_sysroot', '/home/chris/yocto-meta-renew-kirkstone/build/../poky/meta/recipes-core/glib-2.0/glib-2.0_2.72.3.bb:do_populate_sysroot', '/home/chris/yocto-meta-renew-kirkstone/build/../poky/meta/recipes-core/glibc/glibc_2.35.bb:do_populate_sysroot', '/home/chris/yocto-meta-renew-kirkstone/build/../poky/meta/recipes-devtools/cmake/cmake-native_3.22.3.bb:do_populate_sysroot', '/home/chris/yocto-meta-renew-kirkstone/build/../poky/meta/recipes-devtools/gcc/gcc-cross_11.5.bb:do_populate_sysroot', '/home/chris/yocto-meta-renew-kirkstone/build/../poky/meta/recipes-devtools/gcc/gcc-runtime_11.5.bb:do_populate_sysroot', '/home/chris/yocto-meta-renew-kirkstone/build/../poky/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', '/home/chris/yocto-meta-renew-kirkstone/build/../poky/meta/recipes-graphics/drm/libdrm_2.4.110.bb:do_populate_sysroot', '/home/chris/yocto-meta-renew-kirkstone/build/../poky/meta/recipes-graphics/mesa/mesa_22.0.3.bb:do_populate_sysroot', '/home/chris/yocto-meta-renew-kirkstone/build/../poky/meta/recipes-graphics/wayland/libinput_1.19.4.bb:do_populate_sysroot', '/home/chris/yocto-meta-renew-kirkstone/build/../poky/meta/recipes-graphics/wayland/wayland-protocols_1.25.bb:do_populate_sysroot', '/home/chris/yocto-meta-renew-kirkstone/build/../poky/meta/recipes-support/libsoup/libsoup-2.4_2.74.2.bb:do_populate_sysroot', '/home/chris/yocto-meta-renew-kirkstone/build/../poky/meta/recipes-support/libsoup/libsoup_3.0.7.bb:do_populate_sysroot', 'virtual:native:/home/chris/yocto-meta-renew-kirkstone/build/../poky/meta/recipes-devtools/ninja/ninja_1.10.2.bb:do_populate_sysroot', 'virtual:native:/home/chris/yocto-meta-renew-kirkstone/build/../poky/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/home/chris/yocto-meta-renew-kirkstone/build/../poky/meta/recipes-devtools/pkgconfig/pkgconfig_git.bb:do_populate_sysroot', 'virtual:native:/home/chris/yocto-meta-renew-kirkstone/build/../poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/home/chris/yocto-meta-renew-kirkstone/build/../poky/meta/recipes-extended/xz/xz_5.2.6.bb:do_populate_sysroot', 'virtual:native:/home/chris/yocto-meta-renew-kirkstone/build/../poky/meta/recipes-graphics/wayland/wayland_1.20.0.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['wpebackend-fdo', 'wpewebkit', 'glib-2.0', 'glibc', 'cmake-native', 'gcc-cross-arm', 'gcc-runtime', 'quilt-native', 'libdrm', 'mesa', 'libinput', 'wayland-protocols', 'libsoup-2.4', 'libsoup', 'ninja-native', 'patch-native', 'pkgconfig-native', 'pseudo-native', 'xz-native', 'wayland-native', 'gettext-minimal-native', 'libtool-native', 'libwpe', 'libxkbcommon', 'libepoxy', 'wayland', 'linux-libc-headers', 'expat', 'xrandr', 'xorgproto', 'libxfixes', 'libxdamage', 'zlib', 'libx11', 'libxext', 'libxxf86vm', 'libxshmfence', 'libpciaccess', 'libpthread-stubs', 'mpfr-native', 'libmpc-native', 'binutils-cross-arm', 'zlib-native', 'texinfo-dummy-native', 'flex-native', 'gnu-config-native', 'gmp-native', 'zstd-native', 'libxslt', 'sqlite3', 'libpng', 'libpcre', 'freetype', 'gstreamer1.0', 'libxml2', 'gstreamer1.0-plugins-good', 'libjpeg-turbo', 'libwebp', 'gstreamer1.0-plugins-base', 'cairo', 'openjpeg', 'woff2', 'fontconfig', 'harfbuzz', 'libgcrypt', 'libtasn1', 'icu', 're2c-native', 'libgcc', 'libpsl', 'gobject-introspection', 'nghttp2', 'attr-native', 'expat-native', 'libffi-native', 'python3-native', 'libxml2-native', 'mtdev', 'libcheck', 'libevdev', 'systemd', 'util-linux', 'bash-completion', 'libffi', 'ncurses-native', 'bzip2-native', 'curl-native', 'xkeyboard-config', 'libxcb', 'libxrender', 'libxrandr', 'util-macros', 'autoconf-archive', 'xtrans', 'm4-native', 'bzip2', 'python3', 'orc', 'v4l-utils', 'libsm', 'flac', 'lame', 'libgudev', 'wavpack', 'taglib', 'gdk-pixbuf', 'speex', 'libvpx', 'mpg123', 'libcap', 'pulseaudio', 'pango', 'libxv', 'libglu', 'iso-codes', 'libogg', 'alsa-lib', 'libvorbis', 'libtheora', 'libopus', 'opkg-utils', 'pixman', 'lcms', 'tiff', 'brotli', 'libgpg-error', 'libtirpc-native', 'openssl-native', 'util-linux-libuuid-native', 'sqlite3-native', 'gdbm-native', 'readline-native', 'libnsl2-native', 'shadow', 'shadow-sysroot', 'kmod', 'base-files', 'acl', 'base-passwd', 'libseccomp', 'zstd', 'shadow-native', 'ncurses', 'libcap-ng', 'util-linux-libuuid', 'libxcrypt', 'xcb-proto', 'libxdmcp', 'libxau', 'readline', 'gdbm', 'openssl', 'xz', 'libnsl2', 'libtirpc', 'libice', 'shared-mime-info', 'speexdsp', 'libtool', 'dbus', 'libxtst', 'bluez5', 'avahi', 'sbc', 'libatomic-ops', 'ofono', ' [... cut ...]
I am trying to get a more recent version of cog browser because my current version of cog is having problems. The problems are basically having to do with graphics artifacts, occasional freezing, and gstreamer video crashing sometimes. When I have upgraded to newer versions of webkit/cog oddly the graphics artifacts have gotten worse. I am running cog in DRM mode. I have tried a variety of different runtime variables with not much success. The hardware I am using is allwinner R16 SoC with mali GPU. If you have any suggestions on this bigger issue as it relates to cog/webkit pelase let me know. Than kyou.

psaavedra, 2024-11-12:
A dump of `bitbake wpewebkit -e > wpewebkit.txt` could be helpful for checking how WPE is being built.

closed 2025-02-18 by github-actions[bot]