Igalia/meta-webkit #536: Update README.md with additional dependencies.

ferlzc, 2025-02-20:
I notice using meta-webkit the following error:

ERROR: Nothing PROVIDES 'flite' (but /media/ferlzc/ssd480-1/yocto-project-scarthgap/build/../layers/meta-webkit/recipes-browser/wpewebkit/wpewebkit_2.46.6.bb DEPENDS on or otherwise requires it)

the package **flite** belong to meta-openembedded/meta-multimedia 

https://layers.openembedded.org/layerindex/recipe/400936/

it will be nice to document that.

Thanks!

psaavedra, 2025-03-10:
Actually it depends on the following minimum set of layers:

* poky/meta
* poky/meta-poky
* meta-openembedded/meta-oe
* meta-openembedded/meta-python
* meta-openembedded/meta-multimedia

psaavedra, 2025-03-13:
Fixed by https://github.com/Igalia/meta-webkit/pull/541

closed 2025-03-13 by psaavedra